mpogue2/ladybug-mixer #8: TODO: run battery test on the Tencent lithium 9V battery

mpogue2, 2024-04-01:
Test is running now...I expect it will run for about 2 weeks or so.

mpogue2, 2024-05-09:
Done.  It's about 300 hours.

pkbott, 2024-05-09:
Is this continuous use?Ready to try something FUN, UNIQUE and ***@***.*** Square Dance Group:  https://www.facebook.com/groups/squaredanceforfun/

mpogue2, 2024-05-09:
@pkbott Yes.  Here is the graph of my testing, done with 3 different battery technologies.  These were run continuously, while recording the battery voltage every 2 minutes.  I consider a battery "done", when it gets down to 5.5 volts.  The mixer will still run below that voltage, but you'll start to get audio clipping at higher volumes.

This is the reason why I removed the 12V jack (for an external wall-wart style power supply).  When a single alkaline battery lasts 170 hours (that's probably about 50 dances), it doesn't make sense to have to plug it into an outlet.  I worked very hard to keep the power consumption this low, and excellent battery life is the result! :-) :-)

Remember that this is a mixer, and the signal must be sent to a powered speaker (which has the power amplifier inside).
So, the mixer part doesn't actually need much power at all!

I'm also investigating some very inexpensive power amplifiers, perhaps for a future version.  Adding a power amp would definitely require using an external power supply.

[image]

The horizontal red dashed line is where I declare the battery to be "done".  The horizontal dotted green line is where I start notifying the user of low battery, by starting to blink the LED on the front panel.  The lower the battery voltage, the longer the front panel LED will be OFF.  When the battery gets down to the horizontal dotted red line, the front panel LED is on only 1/10 of the time, indicating that battery should be replaced soon.  The LED won't go lower than off 9/10 + on 1/10 of the time, just to make sure that the user sees that the battery is low.  This is all done in software by the microprocessor, which monitors the battery voltage and controls the LED.

closed 2024-10-04 by mpogue2