mpogue2/ladybug-mixer #12: FEATURE: next generation feature list

mpogue2, 2024-04-20:
Now that the first generation is almost complete, it's time to start thinking about the second generation mixer design:

### Possible next gen features:
- 4 mic channels (up from 2)
- 4 remote volume controls (up from 1)
- 4 mute controls (up from 0)
- balance and overall gain controls (up from 0)
- USB interface
  - for power
  - for recording/playback to/from a laptop
  - for control of a music player from a bluetooth dongle
- 2 balanced outputs (up from 1; one for the powered speaker, one for connecting to a Williams transmitter)
- 2 LEDs (up from 1)
  - red peak-reading LED
  - white normal peak-reading LED
  - or a VU-meter style LED array
  - or a small bright display
- 32-bit MCU (e.g. STM32)
  - compression/decoding to/from an SD card
- boost power supply (e.g. 9V -> +/-15V) for increased output
- separate Treble/Bass for Mic channels (e.g. concentric pots)

mpogue2, 2024-05-10:
From Ett:
- Is it possible to add-on a USB plug both the USB C the usb traditional rectangular so we can charge our devices while using this mixer? (This idea would require powering from a wall socket...)

mpogue2, 2025-01-04:
NOTES:  STM32G431CBU6 looks pretty good.  

- WeAct Studio has a nice module that is about $3.  JLCPCB carries it for a couple of dollars.  
- This STM32 is 32-bit, up to 170Mhz, has FPU, has ADC and DAC, lots of GPIO.
- Has [platform.io support](https://github.com/platformio/platform-ststm32/tree/master/examples?utm_source=platformio.org&utm_medium=docs), with several platforms: 
    - Arduino
    - CMSIS (which has DSP functions)
    - libopencm3
    - mbed
    - stm32cube
    - Zephyr

mpogue2, 2025-01-04:
### Various STM32 resources

- Tutorial on [setting up CMSIS-DSP in the STM32Cube IDE](https://community.st.com/t5/stm32-mcus/configuring-dsp-libraries-on-stm32cubeide/ta-p/49637). 

- Tutorial on [getting started with I2C on STM32](https://wiki.st.com/stm32mcu/wiki/Getting_started_with_I2C#Generate_source_code_and_edit_main-c)

- [Datasheet](https://www.st.com/resource/en/datasheet/stm32g431c6.pdf) on the CPU

- CMSIS [main page](https://arm-software.github.io/CMSIS_5/General/html/index.html), includes RTOS, DSP, etc.

mpogue2, 2025-01-04:
- [Getting Started](https://junctionbyte.com/getting-started-with-stm32cubeide-and-stm32g4/) guide

pkbott, 2025-02-03:
1.  Bluetooth capabilities for cordless usage with computer.
2. Amplified option that is cost effective with passive speakers.

mpogue2, 2025-02-03:
@pkbott What kind of passive (non-powered) speakers do you have?  A lot of callers have Yaks, or Hilton Directors, or Linn-Ron's.

pkbott, 2025-02-03:
Just sold my Director and using a Palmetto Pak from Joe Arnold regularly. Ready to try something FUN, UNIQUE and ***@***.*** Square Dance Group:  https://www.facebook.com/groups/squaredanceforfun/

mpogue2, 2025-02-05:
Got another vote for a power amplifier, for people wanting to keep their Yaks, but travel lighter than with an MA-220.

Another suggestion was for a three-position switch to be able to play (L, R) or (R, L) or (L+R, L+R).  Cuers might want to reverse the channels, if their software doesn't do it.

Another vote was to remember to do hearing enhancement.  With what I'm thinking, it would be something a little better than what we have now (Music L/Voice R), e.g. a three-position switch that would do (Music L/Voice R) or (Music R/Voice L) or (Music+Voice/Music+Voice).

pkbott, 2025-02-06:
Interesting comment above about making it modular.  I'm not sure how I feel about that but it may work out well.  I just worry about to many components to carry around personally.  I love the idea of a smaller lighter version of the MA-220.  Of course it should be red too!  I also think that maybe having BlueTooth might be a good option for less cords to the computer.

mpogue2, 2025-03-10:
From Parkers:
Be able to connect two units (with more than a 1/8" cable?) to link them.
One rvc per mic.
Make the mic tone controls more like MA (center position is 4dB treble boost and 8dB bass cut)