mpogue2/ladybug-mixer #25: Feature: Cases for Ladybug LB-102

mpogue2, 2025-01-13:
I'm looking at some from Aliexpress:

[image]

[image]

mpogue2, 2025-01-13:
I also really like the idea of Shadow Foam (aka Kaizen Foam, aka Laminated Foam):

[image]

or 

[image]

mpogue2, 2025-01-13:
Or here:

[image]

pkbott, 2025-01-13:
Personally, I like a case that has room for a microphone and cable as well as maybe a few extra cables/cords that I might need. The perfect case for me would also hold my laptop so it's all in one.